pezi/dart_periphery #3: Can't read more than one byte at time from Serial

mortenboye, 2021-05-31:
Hi, I'm trying to use this library for various I/O in my project, but im struggling to read serial data of arbitrary length.

If I run this code in a periodic timer:
```
SerialReadEvent sre = serial.read(100, 0);
if (sre.count > 0) {
  print("serial has data ${sre.count}");
}
```

I can never read more than one byte at a time, despite sending multiple. Also the byte value doesn't match any of the sent values.
My system is based on a RPI CM4 running flutter-pi and I have dtoverlay=midi-uart0 enabled in the config (mapping 38400bps to 31200 for MIDI).

Any clues as to what is causing this or a good working example?

pezi, 2021-06-01:
1.) What happends, if you increase the timeout parameter?
The original c-periphery documentation says: 
_timeout can be positive for a blocking read with atimeout in milliseconds, zero for a non-blocking read, or negative for a blocking read that will block until length number of bytes are read_

2.) I ported the test unit from the original c-periphery project.
https://github.com/pezi/dart_periphery/blob/main/test/serial_test.dart
I tested this code with the standad UART of the PI and a loop back connection. Perhaps this code is useful for your problem.

mortenboye, 2021-06-02:
Thank you for your suggestions. I found an issue on my hardware so I believe Im back on track. Closing

closed 2021-06-02 by mortenboye