pezi/dart_periphery #13: Unhandled exception when using Docker container

sergeyerofeev, 2022-06-08:
Hi. When launching an application from a Docker container, I get an exception.

Raspberry Pi 3 Model B Rev 1.2

led.dart file for example:
```dart
import 'dart:io';
import 'package:dart_periphery/dart_periphery.dart';

void main() {
  setCPUarchitecture(CPU_ARCHITECTURE.arm64);
  setCustomLibrary('/ex/libperiphery_arm64.so');

  GPIO gpioOut = GPIO(18, GPIOdirection.gpioDirOut);
  
  for (var i = 0; i < 10; ++i) {
    gpioOut.write(true);
    sleep(Duration(seconds: 1));
    gpioOut.write(false);
    sleep(Duration(seconds: 1));
  }
  
  gpioOut.dispose();
}
```
I add a file to the project libperiphery_arm64.so to then copy it to the Docker image.

Creating a Dockerfile:
```dockerfile
FROM dart:stable AS build

WORKDIR /ex

COPY pubspec.* ./
COPY libperiphery_arm64.so .

RUN dart pub get

COPY . .

RUN dart pub get --offline

RUN dart compile exe bin/led.dart -o bin/blink

CMD ["/ex/bin/blink"]
```
Creating an image:
```
docker build -t led .
```
Launching the container:
```
docker run -it --rm led
```
As a result, I get an exception:

[image]

Thank you for your help

pezi, 2022-06-09:
I am no very popular with docker,  but docker is a conatiner environement for encapsulating SW. The exception indicates that the container can not access the hardware.  For direct hardware access you must run the conatainer with more rights. 

`docker run --privileged -d whatever`

[https://stackoverflow.com/questions/30059784/docker-access-to-raspberry-pi-gpio-pins](https://stackoverflow.com/questions/30059784/docker-access-to-raspberry-pi-gpio-pins)

sergeyerofeev, 2022-06-09:
I didn't even think about the lack of rights when accessing hardware. My mistake. 
Thank you for the quick solution of the problem.

closed 2022-06-09 by sergeyerofeev