pezi/dart_periphery #23: Question about sync API

ookami-kb, 2023-09-26:
Is there any specific reason why all the API is sync and waiting is implemented with `sleep` instead of `await`?

I'm using the library with `flutter-pi` and for some reason, it doesn't work great with isolates. The library calls `sleep` throughout the code, making UI laggy.

pezi, 2024-03-01:
I know, the lack of an isolates integration is a problem for flutter-pi.  I am Java developer, and thinking like a Java developer a created a helper class to integrate isolation with help of reflection and annotation some time ago
 
https://github.com/pezi/dart_periphery/blob/ps/0.9.4_isolate_api/lib/src/isolate_helper.dart

But this approach was a hasty action. dart:mirrors doesn't support AOT - therefore no flutter support. This code works only inside CLI based app.

Integreation isolate code by hand works. The only way to add isolate support for this project is to uses code generators like source_gen. In the next time I will do this step.

pezi, 2024-07-27:
New subproject which handles isolates is available
https://github.com/pezi/flutter-pi-sensor-tester