pezi/dart_periphery #24: I2C for PCF8591 A/D  - I might be able to Help

AllyTechEngineering, 2024-02-28:
Hi, great work on the package. I am a Flutter app developer but my previous experiences are in embedded systems such as the STM32 running RTOS and coded in C.

Currently I am using your package to create embedded systems leveraging Flutter for the UI on touch screens on Raspberry Pi 4B. 

After reviewing your code it seems that you are creating IC specific .dart files that are part of your package. 

I may be oversimplifying your code but it seems that the secret sauce is the IC specific files for a particular I2C IC and that more devices can be added if these dart files were developed.

Currently I did not see any A/D IC with I2C in the package and that is a must for embedded Supervisory Control and Data Acquisition (SCADA) systems that I work with. I might be able to help and I can also test on various SBC and give you the results.

Let me know if I can help - btaylor@allytechllc.com or https://www.linkedin.com/in/bob-taylor-mscs-mba/

pezi, 2024-02-29:
Thanks for your offer to help. But this project is still sleeping due obligations at work. 

This project was developed, when I was working for a company with a electrical engineering background. There was the idea to use flutter for SoCs with a terminal. But I changed the job,  only Java and Python now  - no flutter/dard need.

AllyTechEngineering, 2024-02-29:
Thanks for the update.

Bob

pezi, 2025-02-08:
Add offical pcf8591 support to version 0.9.15
https://github.com/pezi/dart_periphery/blob/main/example/i2c_pcf8591.dart

closed 2025-02-08 by pezi