pezi/dart_periphery #26: SPI Example

AllyTechEngineering, 2024-06-18:
I have tested the simple example for SPI using the same BME280 you recommended. 

It is not working - the error is "unknown model"

Linux kernel: Linux 6.6.31+rpt-rpi-v8 aarch64

Setup using raspi-config and added the dtoverlay shown below to config.txt

I have used the https://github.com/rm-hull/spidev-test to do a loop back test on /dev/spidev0.0 and the loop back is working as expected. 
GPIO 9 MISO to GPIO 10 MOSI 

config.txt has these lines included:
dtparam=spi=on,
dtoverlay=spi1-2cs,cs0_pin=18,cs1_pin=17
cs0_spidev=enable
cs1_spidev=enable

ran this on terminal:  ls -l /dev/spi* and spidev0.0 to 1.1 are all showing

Wiring:
GPIO 9 MISO to SDO on module
GPIO 10 MOSI to SDI on module
GPIO 11 SCLK to SCK on module
CS on module to ground (tried it open also)
Tried switching SDI and SDO but still did not work.

Not sure if anyone is monitoring the package but any help would be great.

Here is my code, I added a catch to the try block so I could see the error.
import 'package:dart_periphery/dart_periphery.dart';
import 'package:flutter/foundation.dart';

void main() {
  var spi = SPI(0, 0, SPImode.mode0, 1000000);
  try {
    debugPrint('SPI info:${spi.getSPIinfo()}');
    var bme280 = BME280.spi(spi);
    var r = bme280.getValues();
    debugPrint('Temperature [°] ${r.temperature.toStringAsFixed(1)}');
    debugPrint('Humidity [%] ${r.humidity.toStringAsFixed(1)}');
    debugPrint('Pressure [hPa] ${r.pressure.toStringAsFixed(1)}');
  } catch (e) {
    debugPrint("Error is: $e");
  } finally {
    spi.dispose();
  }
}

pezi, 2024-06-20:
The error rises due the mismatch of sensor id check in the init:

https://github.com/pezi/dart_periphery/blob/main/lib/src/hardware/bme280.dart
```
  void _initialize() {
    // get model
    switch (_readByte(idReg)) {
      case bmp280Id:
        _model = BME280model.bmp280;
        break;
      case bme280Id:
        _model = BME280model.bme280;
        break;
      default:
        throw BME280exception('Unknown model');
    }
```
Due to my lack of soldering skills, my BME280 has contant points, I tested only the SPI loop device.

Is this SPI loop example working for you?
https://github.com/pezi/dart_periphery/blob/main/example/spi_loopback.dart

AllyTechEngineering, 2024-06-20:
Hi, thanks for getting back to me. I too don't solder well but was able to get the header pins with wires solder well to the BME you all recommend from Seeed Studio. I went through the usual stuff in the terminal command line to verify SPI was set up and even tried the other three SPI configurations using the dtoverlay in the config.txt. I built the c-periphery code and used their SPI test code which includes a loop back. I have not used the loopback you show. 
I have a logic analyzer and got the right data from the c-periphery test code. I did go into your code and set up debugPrint to see what model was coming back from _readyByte(idReg), I assume that is where you are getting the model ID? On several BME devices, I was getting D0 hex. However, with nothing connected, I got D0 hex.
When I connected the logic analyzer to the BME and ran your code I did not get anything on MOSI, MISO or SCK so I am wondering if the code is sending out data to request the model number.
I'll try your spi_loopback.dart and see what I get.

AllyTechEngineering, 2024-06-21:
Hi, I tried the spi_loopback.dart and it works fine.

AllyTechEngineering, 2024-06-23:
Do you think that you can test your SPI BME280 and work out what the issues are?

AllyTechEngineering, 2024-06-29:
Hi, if you all don't have time to work on this can you give me some bread crumbs to help me find the root cause? If I fix it what next?

pezi, 2024-06-29:
Sorry for the late answer, because of work I am working on an other location with limited hardware. End of month I am back to my main working location - than I am an able to build up a SPI test environment.

AllyTechEngineering, 2024-07-01:
No worries, thanks for the help. I am building an online class embedded Flutter class that leverages this package and have everything that I need working except SPI. In the future, I want to add more I2C devices (A/D IC for example) and from my read of the software I think that I understand some of the code. How would the logistics of that work out to add it to the package?

AllyTechEngineering, 2024-08-24:
Hi, I was wondering if this was fixed in your latest release.

pezi, 2024-08-27:
I also wondering, because I am not aware, that I changed anything in this area. But I will look, if there was a code change over the time, which was a possible fix.

AllyTechEngineering, 2024-08-27:
I am waiting for the fix concerning the issue I reported on June 18th. Please let me know if you are going to fix the problem. I am working on an online class that will use SPI.

Thanks,
Bob

closed 2025-01-28 by pezi