pezi/dart_periphery #28: PWM issue on pi5

TheSkangel, 2024-10-09:
Hello, I'm currently trying to run the PWM example on my pi5 but I keep getting the following error:

```dart
[ERROR:flutter/runtime/dart_vm_initializer.cc(41)] Unhandled Exception: PWMerrorCode.pwmErrorOpen
#0      _checkError (package:dart_periphery/src/pwm.dart:120)
#1      PWM._openPWM (package:dart_periphery/src/pwm.dart:165)
#2      new PWM (package:dart_periphery/src/pwm.dart:150)
#3      main (package:flutter_pi_demo_app/main.dart:42)
#4      _runMain.<anonymous closure> (dart:ui/hooks.dart:301)
#5      _delayEntrypointInvocation.<anonymous closure> (dart:isolate-patch/isolate_patch.dart:297)
#6      _RawReceivePort._handleMessage (dart:isolate-patch/isolate_patch.dart:184)
```
I'm using the GPIO18 and the following code:
```dart
import 'package:dart_periphery/dart_periphery.dart';
import 'dart:io';

void main() {
setCustomLibrary("/usr/local/lib/libperiphery_arm64.so");
  final pwm = PWM(2, 2);
  try {
    print('start');
    print(pwm.getPWMinfo());
    pwm.setPeriodNs(10000000);
    pwm.setDutyCycleNs(8000000);
    print(pwm.getPeriodNs());
    pwm.enable();
    print("Wait 20 seconds");
    sleep(Duration(seconds: 20));
    pwm.disable();
  } finally {
    pwm.dispose();
    print('end');
  }
 }
```

I tried accessing the provided documentation for pwm in the example but it seems like the provided url was hacked and now points to another website...
Linux kernel `Linux 6.6.31+rpt-rpi-2712 aarch64` 
configurations are `dtoverlay=pwm,pin=18,func=2`.

I also tried using pwm directly with a bash file and there it works just fine:
```
#!/bin/bash
CHANNEL=2
CHIP=pwmchip2
PWM=pwm$CHANNEL

echo $CHANNEL > /sys/class/pwm/$CHIP/export
sleep 1
echo 1000 > /sys/class/pwm/$CHIP/$PWM/period
echo 500 > /sys/class/pwm/$CHIP/$PWM/duty_cycle
echo 1 > /sys/class/pwm/$CHIP/$PWM/enable
```

Does anybody had this issue before or does know how to fix it?

pezi, 2024-10-09:
Thanks for the hijacked PWM website - I wiil replace the URL.

I added your dtoverlay=pwm,pin=18,func=2 defintion to my RPI.

But on my RPI3 (32bit) this definition creates a 
/sys/class/pwm/pwmchip0/pwm0 
device tree

and the demo works.
dart pwm_example.dart 

```
PWM 0, chip 0 (period=0.000000 sec, duty_cycle=nan%, polarity=normal, enabled=false)
10000000
Wait 20 seconds
```

AllyTechEngineering, 2025-01-31:
## My Fix For Model 5 PWM issues - Model 4B solutions in upcoming section.
### Step One Create a Custom dtoverlay
*******************************************************************************************************************
This link has all of the details for creating the dtoverlay that I find works for pwmchip0, pwm0, pwmchip2, pwm0 and pwmchip6 pwm0 (channel 0).
https://gist.github.com/Gadgetoid/b92ad3db06ff8c264eef2abf0e09d569

In a new terminal window:
Use sudo nano and copy this code and save as pwm-pi5-overlay.dts

```
/dts-v1/;
/plugin/;

/{
	compatible = "brcm,bcm2712";

	fragment@0 {
		target = <&rp1_gpio>;
		__overlay__ {
			pwm_pins: pwm_pins {
				pins = "gpio12", "gpio13", "gpio18", "gpio19";
				function = "pwm0", "pwm0", "pwm0", "pwm0";
			};
		};
	};

	fragment@1 {
		target = <&rp1_pwm0>;
		frag1: __overlay__ {
			pinctrl-names = "default";
			pinctrl-0 = <&pwm_pins>;
			status = "okay";
		};
	};
};
```

To compile (copy this at the command line): 
```
dtc -I dts -O dtb -o pwm-pi5.dtbo pwm-pi5-overlay.dts
```
Then to install (copy this at the command line): 
```
sudo cp pwm-pi5.dtbo /boot/firmware/overlays/
```
Comment out any pwm dtoverlays and add the dtoverlay config.txt in /boot/firmware/
```
sudo nano config.txt
```
```
dtoverlay=pwm-pi5
```

### Step Two: Create a udev rule
*******************************************************************************************************************
We need to create a new udev rule that will export the pwm channels at boot. Linux by default does not do this.
```
sudo nano /etc/udev/rules.d/99-pwm.rules
```
Copy and paste the following - if you changed the overlay and the pwmchip_ number change the following to match your changes.

```
ACTION=="add", SUBSYSTEM=="pwm", KERNEL=="pwmchip0", RUN+="/bin/sh -c 'echo 0 > /sys/class/pwm/pwmchip0/export'"
ACTION=="add", SUBSYSTEM=="pwm", KERNEL=="pwmchip2", RUN+="/bin/sh -c 'echo 0 > /sys/class/pwm/pwmchip2/export'"
ACTION=="add", SUBSYSTEM=="pwm", KERNEL=="pwmchip6", RUN+="/bin/sh -c 'echo 0 > /sys/class/pwm/pwmchip6/export'"
```

After saving the file, reload the udev rules.
```
sudo udevadm control --reload-rules && sudo udevadm trigger
[... 123 lines cut ...]
```

Save the file with a name that makes sense to you. For example, pwmchip2_pwm0.sh

To set permissions so you can run the script - make sure the file name is the same as the script.
```
sudo chmod +x pwmchip2_pwm0.sh
```

Use this command in terminal (assuming you are in the same folder) to run the script: 
```
sudo bash ./pwmchip2_pwm0.sh 1000000 500000
```
This sets the period in ns and the duty cycle in ns which results in a 50% duty cycle square wave.
If you set the duty cycle to 0 the PWM is low and if you set it at 100000 it is high.

### Step Five, Test in your Flutter App with dart_periphery
*************************************************************************************************
Using the dtoverlay and udev as shown, I have successfully ran all four PWMs at the same time in my app.
``
pwm0 = PWM(2, 0);
pwm1 = PWM(2, 1);
pwm2 = PWM(2, 2);
pwm3 = PWM(2, 3);
``